mawson-rovers/hestia #66: B4 Cloud 9 not working

LaryssaRaffa, 2023-10-05:
Cloud 9 does not work for B4 (double stacked boards). 
There was a message in the "introduction" tab to update the software. Tried via uts-update command, but got an error message. Tried to run a program, but also got an error message.
Error messages below.
[image]
Control via the dashboard works well.

PS: Did a software update on B1 and it went well.

mryall-mawson, 2023-10-05:
Thanks for reporting this issue, Laryssa. It is a misconfiguration on B4 that you can fix from the terminal. 

Try doing this to link the `.env` file into the Cloud9 path where the terminal is running:

```sh
ln -s ~/.env .
```

After this, if you are still getting an error about missing the UTS_PROGRAM_FILE location, then you’ll need to edit the .env file to add this line:

```toml
UTS_PROGRAM_FILE=/home/debian/uts/uts-programs.toml 
```

Please let me know how you get on!

LaryssaRaffa, 2023-10-05:
Thanks heaps Matt.
Will test it tomorrow and let you know how it goes.
Thanks again!

mryall-mawson, 2023-10-10:
Good news, thanks!

closed 2023-10-10 by mryall-mawson