mawson-rovers/hestia #76: TVac testing scripts

mryall-mawson, 2024-01-18:
Requirements:
* 4 tests required - 2 boards, 2 temps
  * Low temperature hold at -10°C - run heat & cool cycle on top/bottom
  * High temperature hold - ditto

Current testing process at CUAVA:
* `cd UTS`
* `uts-cli test`

Suggested interface for CUAVA:
* `uts-cli tvac-1` - bottom, temp_abort = 80, cool probably not needed?
* `uts-cli tvac-2` - bottom, temp_abort = 125
* `uts-cli tvac-3` - top, temp_abort = 80, cool probably not needed?
* `uts-cli tvac-4` - top, temp_abort = 125

Use existing functionality from uts-run (which loads from UTS_PROGRAM_FILE) to load from tvac-1.toml, tvac-2.toml, etc.

Need to run uts-log in the background, like we do with systemd on test BBB. Could use CubeOS scheduler? Or another way to run system services? Want to capture data before/after uts-cli execution.

Schedule for TVac:
* start `uts-log` as soon as possible after boot
* ensure log output going to `$UTS_LOG_PATH`
* run `uts-cli tvac-1`, etc. manually at specified times

cowen-mawson, 2024-01-19:
From Nick:

> For the TVac test we’ll go for the following:
Full power test
Cut off temp, 85C (measured by TH1)
No high-temp test after all

mryall-mawson, 2024-01-31:
Tested manually via `uts-cli run ...`, and works correctly.

Waiting for scheduler to be working on WS-1 to finalise this.

mryall-mawson, 2024-02-09:
Scheduler is now working. Waiting to receive execution logs to confirm it is running the scripts properly.

mryall-mawson, 2024-02-29:
Fully tested today with the scheduler. We ran both tvac1 and tvac2 schedules, and got the associated data back to UTS.

Some minor remaining issues for CUAVA to fix:
- File transfer fails with spurious error at the end, even though it was successful
- Processes started by the scheduler aren't terminated when the schedule is stopped
- Scheduler periodic tasks don't seem to run periodically (which we are now doing with the `zip` task).

closed 2024-02-29 by mryall-mawson